tesseract-robotics/tesseract #833: Inverse kinematic solutions eliminated when valid redundant solution is available

johnwason, 2022-12-11:
We discovered an interesting issue with inverse kinematics when the joint angle of a solution is outside of the valid range of the joint, but a valid redundant solution is available. For example, on the ABB 1200, the range of Joint 3 is -200 to 70 degrees. If the inverse kinematics are solved using OpwKinematics with an input of -190 on Joint 3, the solver will return 170 degrees since it is being normalized to -180 to 180 degrees for that joint. For the ABB 1200, this will result in no valid solution since 170 degrees is outside of the joint limit. The inverse kinematic solver needs to check if there is a valid redundant solution in these situations.

The offending code is here: https://github.com/tesseract-robotics/tesseract/blob/015a9eb24b4f0eb2f3f6e4ae9f763976c49f4a8a/tesseract_kinematics/core/src/kinematic_group.cpp#L171

This is my solution in a Python project to solve this problem: https://github.com/rpiRobotics/rpi_general_robotics_toolbox_py/blob/cd45cb84ad827b333b102ef1730b2cd86ad6e426/src/general_robotics_toolbox/general_robotics_toolbox_invkin.py#L55

My solution checks for +- one rotation, but I guess it could technically be more than one rotation depending on how the joint range is normalized.

The script that demonstrates the issue (Python): https://github.com/hehonglu123/Motion-Primitive-Planning/blob/5e60119d16d5e1e7cf97f43c984d077ac412f481/toolbox/robots_def.py#L949

Levi-Armstrong, 2022-12-13:
Yea, this is a good catch. We should add an additional check for those that fails to satisfy joint limits check to try and harmonize the solution within the given limits. There may already be a function to do this in Tesseract.

johnwason, 2022-12-29:
@Levi-Armstrong did you find an existing function to solve this?

Levi-Armstrong, 2022-12-31:
Not one that will work directly. I will need to create modified version of the harmonize function to allow harmonizing within a range.

Levi-Armstrong, 2023-01-04:
@johnwason How urgent is this? I am currently in the middle of a refactor for tesseract_qt and hope to have it wrapped up this weekend and then was going to transition to updating this and writing more unit tests for tesseract_kinematics.

Levi-Armstrong, 2023-01-04:
I believe a modified version of [this](https://github.com/tesseract-robotics/tesseract/blob/96831d89e5e046ddff6920446e0b4c1f6b369131/tesseract_kinematics/core/include/tesseract_kinematics/core/utils.h#L492) function can be creates to harmonize within limits.

johnwason, 2023-01-04:
We don't have any deadlines depending on this fix

rjoomen, 2023-08-22:
I believe this can be closed.

Levi-Armstrong, 2023-08-23:
Addressed by #899

closed 2023-08-23 by Levi-Armstrong