tesseract-robotics/tesseract #834: ROS-Independent CI

marip8, 2022-12-16:
Currently the Linux CI jobs use a custom docker image with ROS1 components installed. This is a bit unnecessary and causes problems for downstream ROS-independent or ROS2 builds that use the docker image generated by `tesseract` CI (like [here](https://github.com/tesseract-robotics/tesseract_ros2/pull/35#issuecomment-1353811194)). We should update the CI to be independent of ROS as in https://github.com/tesseract-robotics/boost_plugin_loader/pull/12.

Levi-Armstrong, 2022-12-27:
What about using colcon but avoid using industrial_ci and ros dependendecies?

marip8, 2022-12-28:
Yes, that's what I'm suggesting. That's what we did in the plugin loader library

marip8, 2023-03-07:
I recently opened a PR to add this capability to `industrial_ci`: https://github.com/ros-industrial/industrial_ci/pull/818.

@Levi-Armstrong what do you think about transitioning the builds over to this branch of ICI? It would make the CI builds cleaner and make building downstream CI pipelines for `tesseract_ros/2` and other dependent projects easier to do.

Levi-Armstrong, 2023-03-08:
I don't have an issue with that. Have you received any feedback on your PR?

marip8, 2023-03-08:
Not yet and no comments on the corresponding issue I opened; I'm not sure who to ping for a review

marip8, 2023-03-14:
Closing; addressed in #861

closed 2023-03-14 by marip8