tesseract-robotics/tesseract #844: Building Tesseract URDF on Jammy

ravnicas, 2023-02-01:
I'm having some trouble building Tesseract on Ubuntu 22.04.

```
--- stderr: tesseract_urdf
/usr/bin/ld: error: /usr/lib/x86_64-linux-gnu/libvtkFiltersSources-9.1.so.9.1.0: read: Input/output error
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/tesseract_urdf.dir/build.make:573: libtesseract_urdf.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/tesseract_urdf.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
```

All the (possible) required versions of vtk is also installed, even if i do not quite get, where this one comes from.

```
libvtk9-dev/jammy,now 9.1.0+really9.1.0+dfsg2-3build1 amd64 [installed]
libvtk9-java/jammy,now 9.1.0+really9.1.0+dfsg2-3build1 amd64 [installed]
libvtk9-qt-dev/jammy,now 9.1.0+really9.1.0+dfsg2-3build1 amd64 [installed]
libvtk9.1-qt/jammy,now 9.1.0+really9.1.0+dfsg2-3build1 amd64 [installed,automatic]
libvtk9.1/jammy,now 9.1.0+really9.1.0+dfsg2-3build1 amd64 [installed]
python3-vtk9/jammy,now 9.1.0+really9.1.0+dfsg2-3build1 amd64 [installed,automatic]
vtk9-doc/jammy,jammy 9.1.0+really9.1.0+dfsg2-3build1 all
vtk9-examples/jammy,jammy 9.1.0+really9.1.0+dfsg2-3build1 all
vtk9/jammy,now 9.1.0+really9.1.0+dfsg2-3build1 amd64 [installed,automatic]
```

Within my Tesseract ws is:
```
opw_kinematics 
ros_industrial_cmake_boilerplate 
tesseract 
tesseract_ext 
tesseract_planning
```

I'm hoping for some assistance.

Levi-Armstrong, 2023-02-02:
It is hard to tell from the information posted but Tesseract does not us vtk, but it does use PCL if found. It may be related to an issue the PCL install.

rjoomen, 2023-02-22:
I think it's not a tesseract_urdf issue, as it builds correctly on Jammy for me.

closed 2023-02-22 by Levi-Armstrong