tesseract-robotics/tesseract #848: orocos_kdl dependency

rjoomen, 2023-02-22:
`tesseract_kinematics`, `tesseract_scene_graph`, and `tesseract_state_solver` have the following dependency on orocos_kdl:

```
  <depend condition="$ROS_DISTRO != noetic">orocos_kdl</depend>
  <depend condition="$ROS_DISTRO == noetic">liborocos-kdl-dev</depend>
```

I believe this is outdated, as all newer ROS versions use `liborocos-kdl-dev`. Or if supporting the old dependency is still desired, we should use some code like `>= noetic`, if possible.

rjoomen, 2023-02-22:
Reason I'm asking is that all CI workflows have `orocos_kdl` in `ROSDEP_SKIP_KEYS`, which could be removed if this dependency is fixed.

Levi-Armstrong, 2023-02-22:
Currently the oldest version supported is noetic. I would what ever support noetic and newer versions.

rjoomen, 2023-02-22:
Ugh, Humble has [orocos_kdl_vendor](https://index.ros.org/p/orocos_kdl_vendor), Foxy has [orocos_kdl](https://index.ros.org/p/orocos_kdl), Noetic has nothing, and Melodic has [orocos_kdl](https://index.ros.org/p/orocos_kdl) again. And `liborocos-kdl-dev` is an Ubuntu package, not ROS, but at least that's been available since Bionic. I'll create a PR with `liborocos-kdl-dev`, or do we also have to support Windows?

Levi-Armstrong, 2023-02-22:
We do need to support windows, but I believe this should not impact the windows CI build if you make the proposed change.