tesseract-robotics/tesseract #880: URDF Parsing colon character Issue

marrts, 2023-04-10:
I ran into [this](https://github.com/DavidSpielman/robotiq_85_gripper/pull/1) weird issue when parsing a xacro that included a gazebo tag and then a comment. I narrowed it down to the fact that there is an issue when a comment contains a `:` if it is in between tags.

Examples:
### Produces Error
```
<gazebo>
  <!-- : -->
  ...
</gazebo>
```

```
<link>
  ...
  <!-- : -->
</link>
```

### No Errors
```
<gazebo>
  <!-- random_text -->
  ...
</gazebo>
```

```
<link>
  ...
  <!-- random_text -->
</link>
```
It's also okay if it is outside of any tags or in a xacro macro
```
<!-- : -->
```

```
<xacro:macro name="robot">
  <!-- : -->
  ...
</xacro:macro>
```

Levi-Armstrong, 2023-04-10:
Just to be clear, you generated a urdf from the xacro and comments in the urdf is causing the issue, because  Tesseract does not load xacro files?

marrts, 2023-04-10:
Yes. Passing the resulting URDF to the Tesseract rviz widget gave no issues, but my planning server node would always crash when I had any of the above invalid configurations.

Levi-Armstrong, 2023-04-11:
The error message is also odd having `yaml.scanner.ScannerError` because it is an xml file.

marrts, 2023-04-11:
Agreed, I don't remember seeing that error message every time, but it's clearly pointing to that point. It might be also worth noting I am launching this node from a ROS2 launch file. Are you able to reproduce the error?

Levi-Armstrong, 2023-04-11:
I will test today

gavanderhoorn, 2024-11-04:
By default all parameter values are assumed to be YAML 'compatible' actually, IIRC.

So this is how you avoid that.

marip8, 2024-11-04:
Closing, as this is a ROS2 issue whose solution is addressed in the linked comments

closed 2024-11-04 by marip8