tesseract-robotics/tesseract #885: Getting this error while building on Noetic

alpharomeo911, 2023-05-04:
I'm getting this error especially with tesseract_collision
OS - Ubuntu 20.04 Fresh Install
ROS Noetic
```
Errors     << tesseract_collision:make /home/hsrbtower/hsr_ws/logs/tesseract_collision/build.make.000.log                             
In file included from /usr/include/CL/cl.h:32,
                 from /home/hsrbtower/hsr_ws/src/tesseract/tesseract_collision/vhacd/include/tesseract_collision/vhacd/inc/vhacdVHACD.h:34,
                 from /home/hsrbtower/hsr_ws/src/tesseract/tesseract_collision/vhacd/src/vhacd/VHACD.cpp:47:
/usr/include/CL/cl_version.h:34:104: note: #pragma message: cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 220 (OpenCL 2.2)
   34 | #pragma message("cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 220 (OpenCL 2.2)")
      |                                                                                                        ^
/home/hsrbtower/hsr_ws/src/tesseract/tesseract_collision/bullet/src/tesseract_convex_convex_algorithm.cpp: In member function ‘virtual void tesseract_collision::tesseract_collision_bullet::TesseractConvexConvexAlgorithm::processCollision(const btCollisionObjectWrapper*, const btCollisionObjectWrapper*, const btDispatcherInfo&, btManifoldResult*)’:
/home/hsrbtower/hsr_ws/src/tesseract/tesseract_collision/bullet/src/tesseract_convex_convex_algorithm.cpp:684:26: error: ‘class btConvexPolyhedron’ has no member named ‘initialize2’; did you mean ‘initialize’?
  684 |               polyhedron.initialize2();
      |                          ^~~~~~~~~~~
      |                          initialize
/home/hsrbtower/hsr_ws/src/tesseract/tesseract_collision/bullet/src/tesseract_convex_convex_algorithm.cpp:686:28: error: ‘class btPolyhedralConvexShape’ has no member named ‘setPolyhedralFeatures’; did you mean ‘initializePolyhedralFeatures’?
  686 |               polyhedronB->setPolyhedralFeatures(polyhedron);
      |                            ^~~~~~~~~~~~~~~~~~~~~
      |                            initializePolyhedralFeatures
make[2]: *** [bullet/CMakeFiles/tesseract_collision_bullet.dir/build.make:180: bullet/CMakeFiles/tesseract_collision_bullet.dir/src/tesseract_convex_convex_algorithm.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:378: bullet/CMakeFiles/tesseract_collision_bullet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
cd /home/hsrbtower/hsr_ws/build/tesseract_collision; catkin build --get-env tesseract_collision | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

```

Levi-Armstrong, 2023-05-04:
What version of bullet do you have installed?

alpharomeo911, 2023-05-04:
I'm using v2.87 of bullet.

alpharomeo911, 2023-05-04:
Worked with adding tesseract_ext in the workspace.

Thanks!

closed 2023-05-04 by alpharomeo911

Levi-Armstrong, 2023-05-09:
On 20.04 you should not need tesseract_ext, but need to install the bullet extras using `sudo apt install libbullet-extras-dev`