tesseract-robotics/tesseract #886: FLC:removeLinkHelper segment fault

casiarobot, 2023-05-08:
maybe operate envriment at the same time, such as plan and remove the collision.
segment fault occurs:
    discrete_manager_->removeCollisionObject(name);
There are no issues with bullet

casiarobot, 2023-05-08:
[image]

casiarobot, 2023-05-08:
[image]

casiarobot, 2023-05-08:
[image]

Levi-Armstrong, 2023-05-08:
Can you post the backtrace so I can see where in the fcl manager it occurs?

Levi-Armstrong, 2023-08-17:
Without more information I am unable to investigate. Please reopen if need.

closed 2023-08-17 by Levi-Armstrong