tesseract-robotics/tesseract #904: Is it possible to remove  mesh from rViz?

allicen, 2023-06-14:
Hello.
I can't remove Mesh from rViz.
Should it be deleted after the command tesseract_environment::RemoveLinkCommand is applied, or is this option not provided?
I try to remove the mesh like this:
```
tesseract_environment::Command::ConstPtr renderRemoveLink(std::string link_name) {
  return std::make_shared<tesseract_environment::RemoveLinkCommand>(link_name);
}
bool removeBox(ur5_husky_main::Box::Request &req, ur5_husky_main::Box::Response &res, const std::shared_ptr<tesseract_environment::Environment> &env) {
  Command::Ptr boxLink = renderRemoveLink(req.name);
  if (!env->applyCommand(boxLink)) {
    res.result = "ERROR - remove link box";
    return false;
  }
  res.result = "Remove Box end...";
  return true;
}
```
But it remains visible in rViz. At the same time, when I try to delete the mesh again, I get an error:
```
Warning: Tried to remove Link (table) that does not exist
         at line 1595 in /workspace/src/libs/tesseract/tesseract_environment/src/environment.cpp
```
 
At the same time, the commands tesseract_environment::AddLinkCommand and tesseract_environment::MoveLinkCommand work correctly.

marip8, 2023-06-14:
`tesseract` itself doesn't actually publish environment displays to Rviz. Usually an environment monitor or a ROS plotter from `tesseract_ros` or `tesseract_ros2` publishes the state of the environment. Are you using this environment monitor or plotter class, or do you have your own environment publisher?

allicen, 2023-06-16:
I have updated the tesseract-robotics libraries and now I can delete mesh.

allicen, 2023-06-16:
> Are you using this environment monitor or plotter class, or do you have your own environment publisher?
environment monitor

After the update, command RemoveLinkCommand works, the issue can be closed, thank you!

closed 2023-06-16 by allicen