tesseract-robotics/tesseract #919: tesseract_qt: Cannot locate rosdep definition for [gz-math7]

serboba, 2023-07-25:
Hi,

I'm following the Getting Started installation tutorial (https://tesseract-docs.readthedocs.io/en/latest/_source/intro/getting_started_doc.html) and get the following error at the step:

`rosdep install -y --from-paths ~/tesseract_ws/src --ignore-src --rosdistro noetic`

The error:
`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
tesseract_qt: Cannot locate rosdep definition for [gz-math7]`

and also the step above with:
`wstool init ~/tesseract_ws/src/ ~/tesseract_ws/src/tesseract_planning/dependencies.rosinstall`

there is no file called `dependencies.rosinstall` but I assume `dependencies.repos` is meant.
For that I also opened a pull request:
[](https://github.com/tesseract-robotics/tesseract_docs/pull/18)

Many thanks in advance.

Levi-Armstrong, 2023-07-25:
Here is a [script](https://github.com/tesseract-robotics/tesseract_qt/blob/main/.add-gazebo-ppa) to add the necessary ppa. Then install the following libraries: `libgz-rendering7-dev libgz-common5-dev libgz-math7-dev`

Levi-Armstrong, 2023-07-31:
When using rosdep you need to add the gazebo  packages to the skip keys.

rjoomen, 2023-07-31:
Or run rosdep with option `-r`, this ignores missed dependencies:

    rosdep install --ignore-src --from-paths src -y -r

serboba, 2023-08-03:
It worked, thanks a lot!!

closed 2023-08-03 by serboba