tesseract-robotics/tesseract #923: Improve IK plugin loading error reporting

marip8, 2023-08-11:
Currently when an IK plugin is not successfully created, we are reporting a very generic, unhelpful, and generally inaccurate message:

https://github.com/tesseract-robotics/tesseract/blob/6601a7aa82f48e09b22b78ceeb77fc952a589202/tesseract_kinematics/core/src/kinematics_plugin_factory.cpp#L321

Most of the `create` methods of the existing IK plugin factories throw exceptions with helpful messages when things go wrong, but we are not passing that information along. At the very least, we should report the exception's message. However, I think it would be better to not catch exceptions in `createInvKin` and let the user handle the exception and error messaging. This would be useful for GUI applications where we could report any errors in a widget/dialog box, etc.

Levi-Armstrong, 2023-08-13:
Addressed by #924

closed 2023-08-13 by Levi-Armstrong

Levi-Armstrong, 2023-08-14:
I think that should be fine.