tesseract-robotics/tesseract #925: @Levi-Armstrong  After running `rosdep install --from-paths src --ignore-src -r -y -q`

emfebert, 2023-08-16:
@Levi-Armstrong  After running `rosdep install --from-paths src --ignore-src -r -y -q`

I'm getting

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
tesseract_qt: Cannot locate rosdep definition for [gz-math7]
tesseract_motion_planners: Cannot locate rosdep definition for [trajopt_sqp]
tesseract_examples: Cannot locate rosdep definition for [trajopt_ifopt]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

_Originally posted by @emfebert in https://github.com/tesseract-robotics/tesseract/issues/785#issuecomment-1681121084_

Levi-Armstrong, 2023-08-17:
Does [this](https://github.com/tesseract-robotics/tesseract/issues/919) issue help?

emfebert, 2023-08-18:
Yes, following that it works!

closed 2023-08-18 by emfebert