tesseract-robotics/tesseract #951: Segmentation fault when removing collision objects with FCLDiscreteBVHManager

JayanthSuresh, 2023-10-24:
[image]

Not sure if it is because `FCLDiscreteBVHManager::removeCollisionObject` is trying to remove collision objects from both the `static_manager_` and `dynamic_manger_`.

Levi-Armstrong, 2023-10-24:
Can you provide steps to recreate the issue so I can try and recreate it issue locally?

Example:
1. Create Environment from urdf and srdf
2. Set FCL as the default contact manager
3. apply command which replaces a link which exists
4. ...

JayanthSuresh, 2023-10-25:
High level steps leading to this issue

1. Created the environment from the URDF and SRDF
2. set 'FCLDiscreteBVHManager' as the discrete contact manger
3. Add 2 boxes (0.01, 0.01, 0.01) to the environment using the AddLinkCommand. 
4. Solve for a path using discrete collision checking
5. Remove one of the 2 boxes using RemoveLinkCommand and we will run into segmentation fault

Levi-Armstrong, 2023-10-25:
Awesome, I will create a unit test to recreate the issue and work on a fix.

Levi-Armstrong, 2023-10-25:
Oh!, can you upgrade the latest version?

jayanth-suresh-09, 2023-10-25:
That's a good point. Looks like commit 16837e03d7ec6d8b05cfa35d7c3379589befc199 might address the Issue. I will test with the latest version and confirm if the issue still exists.

jayanth-suresh-09, 2023-10-25:
It wasn't quick to get around the conflicts when upgrading to version 0.20.1. Instead, I just cherry-picked the commit and it solved the issue. Thanks!

closed 2023-10-25 by Levi-Armstrong