tesseract-robotics/tesseract #978: Using trajopt_sqp on Windows fails

Astraaaaaaa, 2024-01-11:
At file: trajopt_sqp\types.h

```
enum class CostPenaltyType
{
  SQUARED,
  ABSOLUTE, // conflict with macro defined in #include <wingdi.h>
  HINGE
};
```

when I use the trajopt_sqp package would encounter with: _error C2143: syntax error : missing '}' before 'constant'_

how do you feel about adding those lines before CostPenaltyType?
```
#if (_WIN32 || _WIN64)
#undef ABSOLUTE
#endif

```

Levi-Armstrong, 2024-01-16:
@johnwason Thoughts?

johnwason, 2024-01-16:
@Astraaaaaaa can you place the trajopt_sqp code in a separate C++ utility file (compilation unit) that doesn't include Windows.h, and then call that utility file from your Windows code? Dealing with this type of issue is a rabbit hole that should be avoidable.