tesseract-robotics/tesseract #983: Why are AVX instructions disabled?

rjoomen, 2024-01-24:
On my Intel Skylake CPU, which has AVX instructions, [the option -mno-avx is passed to the compiler](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_common/cmake/tesseract_macros.cmake#L96), why?

BTW, the result of `uname -p` is x86_64, but that's not enough information to determine AVX support.

rjoomen, 2024-01-24:
Ah, wait, I see: ["Add -mno-avx as compile option to fix Eigen Alignment Issues"](https://github.com/tesseract-robotics/tesseract/commit/4719ab330e30e516927fcf9a5bf22496373155d8).

Levi-Armstrong, 2024-01-24:
Yea, we struggled with a lot and eventually just disabled it. If they would every update the system version of Eigen to the latest this alignment issue should no longer exist.

rjoomen, 2024-01-24:
Unfortunately, updating system Eigen is not the solution. I tested this on Jammy, which has Eigen 3.4.0 (i.e. latest). Problems arise when Tesseract is compiled with AVX, but interfaces through Eigen data types with for example system KDL, which is built without AVX. One specific example is [this line](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_kinematics/kdl/src/kdl_inv_kin_chain_lma.cpp#L84), where a VectorXd is transferred from Tesseract (AVX-aligned) to KDL (not AVX-aligned). When exiting `calcInvKinHelper()` this results in a 'double free or corruption' segfault. For an explanation, [see here](https://stackoverflow.com/a/71667289).

An ugly work-around for this specific case was to resize `kdl_seed` and assign element-wise, i.e. `kdl_seed.data[i] = seed[i]` in a for loop. But then I ran into issues with Boost serialization.

Levi-Armstrong, 2024-01-24:
Yea, it looks like there is still some edge case which overloading the new operator does not handle. We would have to do as you suggested to address these.