tesseract-robotics/tesseract #984: frame is not set to identity

rjoomen, 2024-01-24:
I assume the call to `frame.Identity()` [here](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_kinematics/kdl/src/kdl_utils.cpp#L46) is meant to function like to call to `transform.setIdentity()` a few lines above, but it is a static function and does not change `frame`. I think the correct call should be `frame = frame.Identity();`

Levi-Armstrong, 2024-01-24:
You are correct shown [here](https://github.com/orocos/orocos_kinematics_dynamics/blob/a40783249a21721d49e740824a15de95cd8ddd7d/orocos_kdl/src/frames.inl#L700-L702) but it looks like the default constructor for Vector and Rotation does this so that line can be removed.

rjoomen, 2024-01-24:
Unfortunately not, see [here](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/frames.hpp#L586), the default constructor is empty and `p` and `M` are not initialized by themselves either. Strange decision, compare for example [Twist()](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/frames.hpp#L730), which does properly initialize its members.

Levi-Armstrong, 2024-01-24:
> Unfortunately not, see [here](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/frames.hpp#L586), the default constructor is empty and `p` and `M` are not initialized [...]

The default constructor for `p` and `M` sets `p` to all zeros and `M` to identity. Am I missing something?

rjoomen, 2024-01-24:
Ah, yes, you are right. I was mistakenly believing these had to be explicitly constructed through either an initializer or in the initializer list of the constructor, but that is not the case of course. Actually, I think that the default constructors of both Frame and Twist can be deleted, as the implicit default constructors would have the same functionality, is that correct?

PR #985 (2024-01-25): Remove ineffective frame.Identity() call
Closes #984
closed 2024-01-25 by Levi-Armstrong