tesseract-robotics/tesseract #994: Incorrect octree collision geometry display/representation

marip8, 2024-03-13:
I'm working with a surface mesh that I turn into an octree (sphere outside type) for motion planning collision detection, and I noticed some strange behavior in the visual representation of the octree. It looks a bit like aliasing due to the repeated lines where no voxels are present. This issue also appears for square octree voxel types.

Is this an issue with the actual representation of the collision geometry or just an issue with the rendering of the octree in Rviz?

[image]

rjoomen, 2024-04-30:
Weird. I'm also working with octrees, but haven't encountered this yet.

marip8, 2025-01-27:
Ping @Levi-Armstrong; here is the issue we were discussing earlier today about the octomap visualization