tesseract-robotics/tesseract #1003: Latest Tag

marrts, 2024-05-29:
There hasn't been a tagged version in a while, despite some breaking changes across all the tesseract repos. I recall some talk about tagging a new version after all the forward declaration stuff was merged in, and I think that is done at this point.

Levi-Armstrong, 2024-05-29:
That is correct. I trying to get a few remaining improvements in this week and will create a new release this weekend. 

@marrts, @marip8 and @rjoomen, do you have anything you want to get merged before I cut a new release?

rjoomen, 2024-05-29:
Maybe I have time to finish The tests for the SimplePlanner profiles, would be nice to have that included.

rjoomen, 2024-05-30:
And this: tesseract-robotics/trajopt#405

rjoomen, 2024-05-30:
And this: tesseract-robotics/tesseract#1004

rjoomen, 2024-05-31:
And this: #1008

Levi-Armstrong, 2024-06-11:
I believe everything has been merged.

closed 2024-06-11 by Levi-Armstrong