tesseract-robotics/tesseract #1005: Can TesseractCompoundCollisionAlgorithm and TesseractCompoundCompoundCollisionAlgorithm be removed?

rjoomen, 2024-05-30:
I noticed TesseractCompoundCollisionAlgorithm misses [some old bugfixes](https://github.com/bulletphysics/bullet3/commits/master/src/BulletCollision/CollisionDispatch/btCompoundCollisionAlgorithm.cpp) from `btCompoundCollisionAlgorithm`. I could easily add those to Tesseract, but I noticed that both TesseractCompoundCollisionAlgorithm and TesseractCompoundCompoundCollisionAlgorithm say they could be removed ([here](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_collision/bullet/include/tesseract_collision/bullet/tesseract_compound_collision_algorithm.h#L48) and [here](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_collision/bullet/include/tesseract_collision/bullet/tesseract_compound_compound_collision_algorithm.h#L51)). Is this already the case? I'm not entirely sure what would be required for that. 
In my testing Trajopt ran fine with those removed and even the TesseractConvexConvexAlgorithm removed (although the latter one has modifications important to OMPL planning speed and [should not be removed](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_collision/bullet/include/tesseract_collision/bullet/tesseract_convex_convex_algorithm.h#L63)).

Levi-Armstrong, 2024-05-30:
It will run fine but these were added for performance reason to allow exiting collision checking faster in the event you wanted to collision to terminator of the first contact found. A PR would need to be made against bullet updating the callback to avoid needing these classes. I think the best option is to port the bug fixes and create an MR updating the callback interface.

rjoomen, 2024-05-30:
Thanks for explaining. I can do that, but can you point me to where/how the callback interface has to be updated?
In the meantime I'll port the bugfixes.

rjoomen, 2024-05-31:
Bugfix: #1008