tesseract-robotics/tesseract #1007: conda build action failure when using colcon

johnwason, 2024-05-31:
We are seeing a strange error build error when attempting to build using mambabuild and colcon:

```
usage: colcon [-h] [--log-base LOG_BASE] [--log-level LOG_LEVEL]
              {build,extension-points,extensions,graph,info,list,metadata,test,test-result,version-check}
              ...
colcon: error: unrecognized arguments: -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DBUILD_IPOPT=OFF -DBUILD_SNOPT=OFF -DCMAKE_PREFIX_PATH:PATH=/usr/share/miniconda/envs/tesseract-build/conda-bld/tesseract_1717076257605/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_pl -DTESSERACT_ENABLE_CLANG_TIDY=OFF -DTESSERACT_ENABLE_CODE_COVERAGE=OFF -DTESSERACT_ENABLE_EXAMPLES=OFF -DTESSERACT_BUILD_TRAJOPT_IFOPT=OFF -DSETUPTOOLS_DEB_LAYOUT=OFF -DTESSERACT_ENABLE_TESTING=ON
```

The recipe can be found here: https://github.com/tesseract-robotics/tesseract/blob/master/.github/workflows/conda/recipe/

The colcon command is:

```bash
colcon build --merge-install --install-base="$PREFIX/opt/tesseract_robotics" \
   --event-handlers console_cohesion+ \
   --packages-ignore gtest osqp osqp_eigen tesseract_examples trajopt_ifopt trajopt_sqp \
   --cmake-args -DCMAKE_BUILD_TYPE=Release \
   -DBUILD_SHARED_LIBS=ON \
   -DBUILD_IPOPT=OFF \
   -DBUILD_SNOPT=OFF \
   -DCMAKE_PREFIX_PATH:PATH="$PREFIX" \
   -DTESSERACT_ENABLE_CLANG_TIDY=OFF \
   -DTESSERACT_ENABLE_CODE_COVERAGE=OFF \
   -DTESSERACT_ENABLE_EXAMPLES=OFF \
   -DTESSERACT_BUILD_TRAJOPT_IFOPT=OFF \
   -DSETUPTOOLS_DEB_LAYOUT=OFF \
   -DTESSERACT_ENABLE_TESTING=ON
```

We have been using this recipe for quite a while without issue, and I cannot find anything wrong with the command. I can reproduce the problem locally with the latest version of colcon.

@Tobias-Fischer and @traversaro do you have any suggestions?

Tobias-Fischer, 2024-05-31:
I'd try:
```
colcon build --merge-install --install-base="$PREFIX/opt/tesseract_robotics" \
   --event-handlers console_cohesion+ \
   --packages-ignore gtest osqp osqp_eigen tesseract_examples trajopt_ifopt trajopt_sqp \
   --cmake-args "-DCMAKE_BUILD_TYPE=Release \
   -DBUILD_SHARED_LIBS=ON \
   -DBUILD_IPOPT=OFF \
   -DBUILD_SNOPT=OFF \
   -DCMAKE_PREFIX_PATH:PATH=$PREFIX \
   -DTESSERACT_ENABLE_CLANG_TIDY=OFF \
   -DTESSERACT_ENABLE_CODE_COVERAGE=OFF \
   -DTESSERACT_ENABLE_EXAMPLES=OFF \
   -DTESSERACT_BUILD_TRAJOPT_IFOPT=OFF \
   -DSETUPTOOLS_DEB_LAYOUT=OFF \
   -DTESSERACT_ENABLE_TESTING=ON"
```

johnwason, 2024-05-31:
I tried adding quotes and it did not work either. It passed the quotes to cmake.

traversaro, 2024-05-31:
Not sure what this could be. Are you sure that it is using the colcon from the build environment and not some other colcon in the system?

johnwason, 2024-05-31:
No, the github runner does not have colcon installed so only the conda version should be available.

johnwason, 2024-06-07:
@traversaro do you have any suggestions?

traversaro, 2024-06-07:
Sorry, I am bit out of ideas, perhaps you need to terminate the cmake args with `--` ? See https://github.com/colcon/colcon-core/issues/177

johnwason, 2024-06-11:
I think the problem is this: https://github.com/colcon/colcon-core/pull/635

I manually upgrade to colcon-core 1.16.1 and it appears to work. The current feedstock is still at 1.16.0.

There is an open pull request from autotick to update: https://github.com/conda-forge/colcon-core-feedstock/pull/45

johnwason, 2024-06-11:
This appears to be fixed with the merge of the 0.16.1 pull request in the feedstock. There are some new unrelated conda errors that I will need to fix before the tests will pass.

closed 2024-06-11 by johnwason