tesseract-robotics/tesseract #1053: CompoundMesh does not work with FCL

rjoomen, 2024-10-21:
[CompoundMesh ](https://github.com/tesseract-robotics/tesseract/pull/1046) does not work with the FCL collision checker. I suppose it is because for Bullet, one CompoundMesh is added as one compound object, whereas for FCL, one CompoundMesh is added a multiple objects.

I've taken a look, it seems FCL does not support compound objects, apart from [BHVModel](https://github.com/flexible-collision-library/fcl/blob/master/include/fcl/geometry/bvh/BVH_model.h), which only supports meshes or pointclouds as submodels.

And for reference, neither does HPP-FCL: [#340](https://github.com/humanoid-path-planner/hpp-fcl/issues/340), [#448](https://github.com/humanoid-path-planner/hpp-fcl/issues/448).

rjoomen, 2024-10-21:
I checked the code indeed. There is a bug, I tested adding some meshes separately (within a link) or as a CompoundMesh, and for Bullet this works for both, but for FCL the CompoundMesh version crashes. And with my HPP-FCL version (mostly similar to the FCL version) it is the same. I did not check in detail what went wrong, but CompoundMesh is the cause.

Levi-Armstrong, 2024-10-21:
I thought this was covered by a unit test but I will look into it.

Levi-Armstrong, 2024-10-27:
There is a unit test which does confirm this is working but may not be testing everything. Do you have a backtrace or could you add a unit test which triggers the failure?

rjoomen, 2024-10-27:
I'll take a look tomorrow. I'm not really using FCL, so I just found this by accident.

Levi-Armstrong, 2024-12-10:
@rjoomen Did you get a chance to look into this?

rjoomen, 2024-12-11:
I just did. It was quite hard to find, but the problem is this:

In the [COW](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_collision/fcl/src/fcl_utils.cpp#L338) constructor, a number of shapes and shape_poses is added. Before, each one shape was added as one COW, and thus had a matching shape_pose (and `shapes_.size() == shape_poses_.size() == collision_objects_.size()`). With a CompoundMesh, one shape is added as multiple COWs, and `shape_poses_.size() != collision_objects_.size()`. So when for example [setCollisionObjectsTransform](https://github.com/tesseract-robotics/tesseract/blob/master/tesseract_collision/fcl/include/tesseract_collision/fcl/fcl_utils.h#L119) is executed for a CompoundMesh COW, it runs through, say, 5 objects, but only 1 object gets a valid shape_pose_.

I have submitted [a PR](https://github.com/tesseract-robotics/tesseract/pull/1080) to fix this, but please take a good look if this is the best solution.

rjoomen, 2024-12-11:
The tests didn't pick this up as no `setCollisionObjectsTransform` is performed on a `CompoundMesh`.

PR #1083 (2024-12-18): Fix use of compound mesh in FCL
Fix #1053 

@rjoomen After looking through the code I went a slight different route to solve another issue with get the correct shape index. Do you mind testing this?
closed 2024-12-18 by Levi-Armstrong