tesseract-robotics/tesseract #1059: Consider replacing Tesseract URDF version property with XML namespace

marip8, 2024-11-06:
We should consider using a URDF namespace rather than a version tag for `tesseract`-specific URDF elements. URDF already has a version tag, so it's a bit confusing (and feels unnecessary) to specify an additional `tesseract_version` property.

Here's an example of what this might look like:

```xml
<robot name="my_robot" xmlns:tesseract="http://github.com/tesseract-robotics">
...
<collision>
  <geometry>
    <tesseract:convex_mesh file="" convert=""/>
  </geometry>
</collision>
```

Levi-Armstrong, 2024-12-10:
@marip8 Do you have time to work on this? I think this might be good to get into the next release.

Levi-Armstrong, 2025-07-05:
@marip8 Can this be closed?

marip8, 2025-07-07:
Closing; addressed by #1081

closed 2025-07-07 by marip8