tesseract-robotics/tesseract #1088: Expose access to IK solver tip frames

marip8, 2024-12-16:
### Is your feature request related to a problem? Please describe.

It would be benficial for `KinematicGroup` to expose access to just the tip frame(s) and base frame defined by the IK solver, rather than all possible tip and working frames

### Describe the solution you'd like

```c++
std::vector<std::string> KinematicGroup::getIKSolverTipFrameNames() const
{
  return inv_kin_->getTipLinkNames();
}

std::string KinematicGroup::getIKSolverBaseFrameName() const
{
  return inv_kin_->getBaseLinkName();
}
```

### Describe alternatives you've considered

Access to the IK solver interfaces directly is limited through the current `Environment` API, so I don't think this is currently possible

Levi-Armstrong, 2024-12-17:
Should we just expose a getter to return the IKSolver?

marip8, 2024-12-18:
Yeah, that's probably a better approach

closed 2024-12-21 by Levi-Armstrong