tesseract-robotics/tesseract #1117: Investigate Collision Shape caching

Levi-Armstrong, 2025-01-29:
see PR #1116

rjoomen, 2025-01-29:
I didn't test this yet, but I noticed that you also removed `shape_index` in a number of places in the [original PR](https://github.com/tesseract-robotics/tesseract/pull/1115). Are you sure it doesn't have to do with these modifications, instead of with the cache?

Levi-Armstrong, 2025-01-30:
I don't think so because they are still removed with me just commenting out the cache line and everything works again.

Levi-Armstrong, 2025-01-30:
I was able to track down the issue and will need to rework how the caching works. The issue is that bullet does not manager the collision shape resource and it is on us to manage it. In cases like the octomap and detailed mesh it create subshapes which get stored in the collision object wrapper class but we are not caching these and were only caching the parent object so under certain cases these were subshapes were going out of scope causing the crash I was seeing. I will need to rework this caching approach and also going to make this a member of the contact manager so things will get cleaned up more frequently. I have concerns with the global cache if used with a planning server which allows loading jobs each with a different environment which would eventually run out of memory.

Levi-Armstrong, 2025-02-11:
Addressed by #1120

closed 2025-02-11 by Levi-Armstrong