tesseract-robotics/tesseract #1131: Add unit test to ensure compatibility with `urdfdom` XSD

marip8, 2025-03-19:
Per [this comment](https://github.com/tesseract-robotics/tesseract/issues/1081#issuecomment-2737657921_], we should add a unit test to ensure the customizations added to the `tesseract` URDF parser comply with the `urdfdom` XSD.

marip8, 2025-03-19:
@rjoomen do you know if this is can be an automated process? Or would we need to parse the `urdfdom` XSD file and make sure we don't have conflicting elements?

rjoomen, 2025-03-20:
Something like this?
- Have some Tesseract URDF files for testing in this repo that showcase all possible Tesseract URDF features.
- Clone the XSD from the urdfdom repo in the testing Github workflow.
- Create test cases validating the URDFs against the XSD using some C++ library that supports this. This introduces an extra dependency, but only for testing.

omnilink-tech, 2026-08-28:
I ran this against the current `master` of both repos, and the answer splits into
three independent parts — only one of which is a tesseract bug.

Setup: `ros/urdfdom@master:xsd/urdf.xsd` (14,567 B) against all 8
`support/urdf/*.urdf`, validated with libxml2 via lxml.

**1. As shipped, 8 of 8 fail, and all 8 fail on the same thing:**

```
Element 'robot', attribute '{https://github.com/tesseract-robotics/tesseract}make_convex':
  ... not allowed
```

`urdf.xsd` contains **zero** occurrences of `anyAttribute`, so it structurally
rejects every foreign-namespace attribute, not just yours. That is the direct
consequence of #1081 meeting a schema that has no extension point. It is a
urdfdom-side gap rather than something tesseract can fix in its own files —
the schema would need `<xs:anyAttribute namespace="##other" processContents="lax"/>`
on the elements you extend. Worth raising there; ros/urdfdom#230 is actively
versioning the schema toward 1.1/1.2 right now, so the timing is good.

**2. Strip only those attributes, and 5 of 8 still fail — those are real.**

`<material>` declared inside `<collision>`. The URDF spec puts `<material>` under
`<visual>` only; a collision element takes `origin` and `geometry`. 18
occurrences:

| file | lines |
|---|---|
| `abb_irb2400.urdf` | 41 |
| `car_seat_demo.urdf` | 21, 41, 57, 73, 89, 105, 121, 135, 348, 364, 387, 403, 419 |
| `online_planning_example.urdf` | 41, 365, 409 |
| `scene_graph_example.urdf` | 41 |

Plus one that is plainly a typo — `puzzle_piece_workcell.urdf` line 226 carries a
stray escaped `&gt;` as a text node between `</geometry>` and `</visual>`:

```xml
      </geometry>
      &gt;

    </visual>
```

`boxbot.urdf`, `iiwa7.urdf` and `lbr_iiwa_14_r820.urdf` are clean once the
extension attributes are permitted.

**3. Careful — the oracle is itself non-conformant, and a naive test will go red
for the wrong reason.**

urdfdom's own reference document fails urdfdom's own schema:

```
xsd/autogenerated.urdf  ->  L2: Element '{http://www.ros.org}robot':
                            No matching global declaration available for the validation root
```

The document declares `xmlns="http://www.ros.org"` while the XSD has no
`targetNamespace`. So any test that clones the XSD from upstream needs to either
strip the default namespace before validating or pin a known-good schema
revision. I would not have guessed that from reading either file, and it is the
kind of thing that makes a new CI job look like a tesseract regression on its
first run.

I did not open a PR — we do not open those from our side. I also deliberately did
not attach a bulk patch for part 2: my first automated pass at removing those 18
`<material>` blocks produced malformed XML on `car_seat_demo.urdf`, and the
edits are trivial enough that you would rather make them than review a diff of
mine. The line numbers above are exact.

Happy to write up part 1 as an issue on `ros/urdfdom` with this evidence if that
would help — it seems more likely to move there than here.