tesseract-robotics/tesseract #1154: When running the compiled `tesseract_urdf_load_urdf_example.exe` on a Windows system, the terminal becomes blocked and unresponsive.

CodingHzx, 2025-05-09:
### Version

0.30.0

### OS Version

Windows (MSVC 2022)

### Describe the bug

hi,
After compiling the Tesseract packages on a Windows system, the build completes successfully. However, when running the `tesseract_urdf_load_urdf_example.exe` executable, the terminal remains stuck and unresponsive. I would like to ask what could be causing this issue. All other `*_example.exe` executables in Tesseract run and produce output normally.

[image]

### To Reproduce

As mentioned above, when running the compiled `tesseract_urdf_load_urdf_example.exe` on a Windows system, the terminal becomes stuck and unresponsive.

### Expected behavior

i hope normal log output

### Relevant log output

```shell

```

Levi-Armstrong, 2025-05-13:
@johnwason Any ideas?

johnwason, 2025-05-14:
Try setting the environmental variable `OMP_NUM_THREADS=1` There is a problem with one of the dependencies that can cause a deadlock when initializing the DLL due to OMP.

CodingHzx, 2025-05-24:
Thank you. With some adjustments, I have successfully deployed and run it on Windows. The results are as follows.

[image]

closed 2025-05-27 by CodingHzx